diffblue/hw-cbmc #35: Problems compiling current EBMC

mgudemann, 2020-05-26:
Hi,
I tried building ebmc using `gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)`

There were two problems:
- unused variable `constraint` causing an error because of the compiler settings in `hw-cbmc/hw_cbmc_parse_options.cpp`
```
  // the 'extra constraints'
  if (!constraints.empty()) {
    log.status() << "converting constraints" << messaget::eom;

    for (const auto &constraint : constraints) {
      // TODO : include the extra constraints using the new custom verifier
      // interface
    }
  }
```
  this is easy to solve of course

- undefined references for all the `ID_XXX` in `hw_cbmc_irep_ids.h`
  for this I copied the contents of the file to CBMC's `src/util/irep_ids.def`

But my guess is that this is not the intended way to do this.
What I am missing?

dh73, 2020-09-18:
A step that seems to be missing and is not very obvious is, inside `lib/cbmc` do a `make -C src minisat2-download`, then all definitions are stored there. 

But what doesn't make much sense is that your build was able to reach that point if the minsat2 files was missing, because other errors such as missing references to `Minisat::vec` should have appeared too.

kroening, 2023-10-01:
@mgudemann May I ask you to try whether this works for you now?

mgudemann, 2023-10-03:
no problem compiing a clean checkout of 4961ca57318128b609ca5625d84f91a266b87d46 using `gcc version 9.4.0`
The only issue was that the `README` says
```
download minisat: `$> make -C lib/cbmc minisat2-download`
```
This should be
```
 download minisat: `$> make -C lib/cbmc/src minisat2-download`
```

PR #89 (2023-10-03): edits to README.md
Closes #35
commit 431d367f (2023-10-03): README: fix minisat instructions

Closes #35
closed 2023-10-03 by tautschnig