diffblue/hw-cbmc #274: issue in implication operator

hayat10x, 2023-12-11:
When I execute example3.sv which was given by ebmc, it give me the error "file example3.sv line 14: syntax error before `|->'
PARSING ERROR". Are implication operators supported in ebmc or there is any other mistake I am doing? 
[image]

tautschnig, 2023-12-11:
Could you please let us know how you obtained/built ebmc, and what operating system you are using? The following works fine for me (on Ubuntu 20.04, with ebmc built from source at revision f067e32ed4e):
```
$ ebmc regression/ebmc/example3/example3.sv --module counter --bound 10
Parsing regression/ebmc/example3/example3.sv
Converting
Type-checking Verilog::counter
Using MiniSAT 2.2.1 with simplifier
Generating Decision Problem
Properties
Solving with propositional reduction
Checking counter.property.1
Runtime Post-process: 5.732e-06s
SAT checker: instance is UNSATISFIABLE
UNSAT: No counterexample found within bound
Checking counter.property.2
SAT checker: instance is SATISFIABLE
SAT: counterexample found

** Results:
[counter.property.1] always (counter.enable && counter.state[0] |-> (##1 !counter.state[0])): PROVED up to bound 10
[counter.property.2] always counter.state != 3: REFUTED
```

hayat10x, 2023-12-11:
I built ebmc by follwoing the steps mentioned in compiling file and my os is Ubuntu 22.04.3.

hayat10x, 2023-12-11:
after cloning the hw-cbmc repository I did steps as follows:
1. git submodule init; git submodule update
2. apt-get install g++ gcc flex bison make git curl patch
3. apt-get install openjdk-8-jdk maven jq
4. git clone https://github.com/diffblue/cbmc cbmc-git
5. cd cbmc-git
6. make -C src minisat2-download
7. make -C src
8. make -C jbmc/src setup-submodules
9. make -C jbmc/src
10. export PATH=/full/path/hw-cbmc/src/ebmc:${PATH}

hayat10x, 2023-12-11:
I am able to build ebmc but implication operators are not working. Assertions work fine without implication.

kroening, 2023-12-11:
May I suggest following the steps in the README.md file, as follows:

- initialize and update the CBMC submodule: `$> git submodule init; git submodule update`
- download minisat: `$> make -C lib/cbmc/src minisat2-download`
- build EBMC: `$> make -C src` (this also builds the CBMC submodule)

I'll fix up the COMPILING file.

kroening, 2023-12-11:
> I am able to build ebmc but implication operators are not working. Assertions work fine without implication.

Can you paste `example3.sv` here?

hayat10x, 2023-12-11:
> > I am able to build ebmc but implication operators are not working. Assertions work fine without implication.
> 
> Can you paste `example3.sv` here?

module counter(
  output [7:0] out,
  input enable, input clk);

  reg [7:0] state;
  assign out = state;

  initial state = 0;

  always @(posedge clk)
    if(enable)
      state = state + 1;

  assert property (enable && state[0] |-> ##1 !state[0]);
  assert property (state!=3);

endmodule

hayat10x, 2023-12-11:
> * make -C lib/cbmc/src minisat2-download

I followed README.md an Error pops up during make -C src.

kroening, 2023-12-11:
This does parse here just fine; perhaps try a `make clean` in the `src` subdirectory, and then compile again?

kroening, 2023-12-11:
Also, do check if you've maybe got an older version of ebmc somewhere in your `PATH`.

hayat10x, 2023-12-11:
> Also, do check if you've maybe got an older version of ebmc somewhere in your `PATH`.

I just deleted everything and install again with readme.md steps. Now I got no error but this time ebmc didn't build.

kroening, 2023-12-11:
> I just deleted everything and install again with readme.md steps. Now I got no error but this time ebmc didn't build.

Ok, that might be a reason to improve the compilation instructions. What's the error you are getting?

hayat10x, 2023-12-11:
> p

Command 'ebmc' not found, did you mean:
  command 'cbmc' from deb cbmc (5.12-5)
  command 'jbmc' from deb cbmc (5.12-5)
Try: sudo apt install <deb name>

hayat10x, 2023-12-11:
I am trying to install with new instruction added in compile file.

hayat10x, 2023-12-11:
> > I just deleted everything and install again with readme.md steps. Now I got no error but this time ebmc didn't build.
> 
> Ok, that might be a reason to improve the compilation instructions. What's the error you are getting?

sorry with readme.md steps I mean old steps that was mentioned. I will install again with new steps that you changed in compilation file.
Thanks

hayat10x, 2023-12-11:
Thanks @kroening and @tautschnig for your help. It's working now.