diffblue/hw-cbmc #635: Cover property erroneously getting refuted under bounded model checking

ShashankVM, 2024-08-14:
Here is my project:

https://github.com/ShashankVM/hamming_encoder_decoder_bmc

The cover property here on line 65 is getting refuted. https://github.com/ShashankVM/hamming_encoder_decoder_bmc/blob/master/dut.sv#L65

Run command: 
`ebmc -D FORMAL --bound 100 --top dut dut.sv cc_encoder.sv piso.sv lfsr_encoder.sv cc_decoder_ht.sv lfsr_decoder.sv buffer_reg.sv error_correction_and_detection.sv channel.sv --reset reset==1 --vcd wave.vcd`

It is expected to be covered as there is no constraint on the input signal message that prevents it from being zero. I've not been successful in generating a waveform via VCD file that can visualize the design behaviour using the HW-CBMC tool.

k-induction and bdd also fails.

kroening, 2024-08-30:
Can you confirm whether this is now fixed?

ShashankVM, 2024-08-31:
Thank you, Dr. Daniel. Yes it's fixed for the bounded model checking. K-induction and BDD does not work.

kroening, 2024-09-01:
cover properties are existential. k-induction or BDDs can only refute them, not prove them.

kroening, 2024-09-01:
Closing as fixed

closed 2024-09-01 by kroening

ShashankVM, 2024-09-07:
Hi @kroening Sir, both k-induction and BDD fail to prove the assertions. I added an ifdef so that the cover properties aren't compiled when proving k-induction and BDD.

K-induction run command: 
`ebmc -D FORMAL --k-induction --top dut dut.sv cc_encoder.sv piso.sv lfsr_encoder.sv cc_decoder_ht.sv lfsr_decoder.sv buffer_reg.sv error_correction_and_detection.sv channel.sv --reset reset==1`

Here is the output for k-induction:
```
[dut.ASSUME_STABLE_MESSAGE] always (disable iff (dut.reset) $fell(dut.encoder_ready) or !dut.encoder_ready |-> $stable(dut.message)): FAILURE: property unsupported by k-induction
[dut.ASSUME_STABLE_ERROR_INJECT] always (disable iff (dut.reset) $fell(dut.encoder_ready) or !dut.encoder_ready |-> $stable(dut.error_inject)): FAILURE: property unsupported by k-induction
[dut.ASSUME_STABLE_ERROR_POSITION1] always (disable iff (dut.reset) $fell(dut.encoder_ready) or !dut.encoder_ready |-> $stable(dut.error_pos1)): FAILURE: property unsupported by k-induction
[dut.ASSUME_STABLE_ERROR_POSITION2] always (disable iff (dut.reset) $fell(dut.encoder_ready) or !dut.encoder_ready |-> $stable(dut.error_pos2)): FAILURE: property unsupported by k-induction
[dut.ASSUME_VALID_ERROR_POSITION1] always (disable iff (dut.reset) dut.error_pos1 < 3'b111): ASSUMED
[dut.ASSUME_VALID_ERROR_POSITION2] always (disable iff (dut.reset) dut.error_pos2 < 3'b111): ASSUMED
[dut.ASSERT_RX_EQUAL_TX_NO_INJECTION] always (disable iff (dut.reset) dut.rx_valid && !dut.error_inject_reg |-> dut.tx_final == dut.rx): INCONCLUSIVE
[dut.ASSERT_RX_EQUAL_TX_SINGLE_ERROR_WITH_INJECTION] always (disable iff (dut.reset) dut.rx_valid && $past(dut.error_pos1) == $past(dut.error_pos2) && dut.error_inject_reg |-> dut.tx_final == dut.rx): INCONCLUSIVE
[dut.ASSERT_RX_NOT_EQUAL_TX_FOR_DOUBLE_ERROR] always (disable iff (dut.reset) dut.rx_valid && $past(dut.error_pos1) != $past(dut.error_pos2) && dut.error_inject_reg |-> dut.tx_final != dut.rx): INCONCLUSIVE
[dut.ASSERT_ENCODER_LATENCY] always (disable iff (dut.reset) $rose(dut.encoder_ready) |-> (##7 dut.encoder_ready)): FAILURE: property unsupported by k-induction
[dut.ASSERT_DECODER_LATENCY] always (disable iff (dut.reset) $rose(dut.rx_valid) |-> (##7 dut.rx_valid)): FAILURE: property unsupported by k-induction
[dut.ASSERT_ERROR_DET_IF_ERROR_INJECT] always (disable iff (dut.reset) dut.rx_valid && dut.error_inject_reg |-> dut.error_det): INCONCLUSIVE
[dut.ASSERT_ERROR_DET_IMPLIES_ERROR_INJECT] always (disable iff (dut.reset) dut.error_det |-> dut.error_inject_reg): INCONCLUSIVE
[dut.ASSERT_RX_EVENTUALLY] always (disable iff (dut.reset) s_eventually dut.rx_valid): FAILURE: property unsupported by k-induction
```

For BDD:
Run command: 
`ebmc -D FORMAL --bdd --top dut dut.sv cc_encoder.sv piso.sv lfsr_encoder.sv cc_decoder_ht.sv lfsr_decoder.sv buffer_reg.sv error_correction_and_detection.sv channel.sv --reset reset==1`

```
vcd wave.vcd
Parsing dut.sv
Parsing cc_encoder.sv
Parsing piso.sv
Parsing lfsr_encoder.sv
Parsing cc_decoder_ht.sv
Parsing lfsr_decoder.sv
Parsing buffer_reg.sv
Parsing error_correction_and_detection.sv
Parsing channel.sv
Converting
Type-checking Verilog::buffer_reg
Type-checking Verilog::error_correction_and_detection
file error_correction_and_detection.sv line 7: Making error_correction_and_detection.error_pos a wire
Making error_correction_and_detection.error_det a wire
Making error_correction_and_detection.code_out a wire
Type-checking Verilog::lfsr_decoder
Type-checking Verilog::cc_decoder_ht
file cc_decoder_ht.sv line 17: Making cc_decoder_ht.next a wire
file cc_decoder_ht.sv line 11: Making cc_decoder_ht.clear_2 a wire
file cc_decoder_ht.sv line 11: Making cc_decoder_ht.clear_1 a wire
file cc_decoder_ht.sv line 11: Making cc_decoder_ht.valid_code_in a wire
file cc_decoder_ht.sv line 10: Making cc_decoder_ht.code_in_2 a wire
file cc_decoder_ht.sv line 10: Making cc_decoder_ht.code_in_1 a wire
Making cc_decoder_ht.code_out a wire
Making cc_decoder_ht.error_det a wire
Making cc_decoder_ht.valid_out a wire
file cc_decoder_ht.sv line 11: Making cc_decoder_ht.error_det_signal a wire
file cc_decoder_ht.sv line 13: Making cc_decoder_ht.buffer_out a wire
file cc_decoder_ht.sv line 15: Making cc_decoder_ht.syndrome_val a wire
file cc_decoder_ht.sv line 15: Making cc_decoder_ht.syndrome_val_1 a wire
file cc_decoder_ht.sv line 15: Making cc_decoder_ht.syndrome_val_2 a wire
Type-checking Verilog::lfsr_encoder
Type-checking Verilog::piso
Making piso.piso_out a wire
Type-checking Verilog::cc_encoder
Making piso(4).piso_out a wire
Making cc_encoder.valid a wire
file cc_encoder.sv line 13: Making cc_encoder.next a wire
Making cc_encoder.ready a wire
Making cc_encoder.code_out a wire
file cc_encoder.sv line 9: Making cc_encoder.lfsr_reg a wire
[... 36 lines cut ...]
ebmc(+0x32e074) [0x635ea8892074]
ebmc(+0x32e8bf) [0x635ea88928bf]
ebmc(+0x32de9d) [0x635ea8891e9d]
ebmc(+0x32e1e8) [0x635ea88921e8]
ebmc(+0x32e074) [0x635ea8892074]
ebmc(+0x32e8bf) [0x635ea88928bf]
ebmc(+0x32de9d) [0x635ea8891e9d]
ebmc(+0x3304e6) [0x635ea88944e6]
ebmc(+0x331ed2) [0x635ea8895ed2]
ebmc(+0xa6909) [0x635ea860a909]
ebmc(+0xa7324) [0x635ea860b324]
ebmc(+0xdde6d) [0x635ea8641e6d]
ebmc(+0xb7943) [0x635ea861b943]
ebmc(+0x8fb7f) [0x635ea85f3b7f]
ebmc(+0x8dc59) [0x635ea85f1c59]
/lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7912b3e2a1ca]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7912b3e2a28b]
ebmc(+0x98015) [0x635ea85fc015]

--- end invariant violation report ---
Aborted (core dumped)

```

As you can see, the COVER properties aren't compiled here, still not able to execute the full proof. Let me know if you want me to raise a separate Issue for this.