diffblue/hw-cbmc #771: Using indexed part-select in Verilog file causes syntax error

zputrle, 2024-10-16:
If I run the following example with `ebmc examples/test_part_select.v --top test_part_select --bound 10`
```
module test_part_select(input clk, input [31 : 0] x);

    reg [2:0] x_r;

    always @(posedge clk) begin
        x_r = x[10 -: 3];
    end

endmodule
```
I get the following output
```
Parsing examples/test_part_select.v
file examples/test_part_select.v line 6: -: is a System Verilog operator before '-:'
file examples/test_part_select.v line 6: syntax error before '3'
```
This problem does not occur if I force  EBMC to parse the example as SystemVerilog using `--systemverilog` flag.

It seems that the indexed part-select is not supported by EBMC in scope of Verilog even though it is specified in Verilog 2005 specification.

kroening, 2024-10-16:
Indeed; please try the fix in #772.

zputrle, 2024-10-17:
The fix solves the problem. Thank you.

commit 0cd3d4a3 (2024-10-16): Verilog: indexed part select is Verilog 2005

The Verilog 2005 standard already allows indexed part select expressions
(Sec. 5.2.1).  This re-classifies the +: and -: tokens from SystemVerilog
to Verilog.

Fixes #771.
commit c193387f (2024-10-16): Verilog: indexed part select is Verilog 2005

The Verilog 2005 standard already allows indexed part select expressions
(Sec. 5.2.1).  This re-classifies the +: and -: tokens from SystemVerilog
to Verilog.

Fixes #771.
PR #772 (2024-10-17): Verilog: indexed part select is Verilog 2005
The Verilog 2005 standard already allows indexed part select expressions (Sec. 5.2.1).  This re-classifies the `+:` and `-:` tokens from SystemVerilog to Verilog.

Fixes #771.
closed 2024-10-17 by tautschnig